googlecreativelab/coder #94: blinky_lights "SERVERLOG: "EPERM""

bmjbmj, 2015-02-24:
The demo program submitted by: Coder Team produce a error response "SERVERLOG: "EPERM" when the button is pressed in the browser the LED do not light on the R.Pi side. I have debugged it as far as that the problem is white the call to "ledDevice.set(val);" The opposite direction do work. And if I use Python (as sudo) the I/O on pin 18 works fine.

If the code used is:
console.log( 'OK'+val);
ledDevice.set(val);

The console reads:
"SERVERLOG: "OK1"" index.js:79
"SERVERLOG: "EPERM""
When the button is pressed

If the line is commented out:
console.log( 'OK'+val);
//ledDevice.set(val);

The console reads:
"SERVERLOG: "OK1""

The init of the I/O looks OK but I guess something goes wrong never the less??? Dose Coder run white the right privileges???

"socket connected" index.js:69
"Connected with id: 4jIA0hgJal5RfsMtr0z8LjYmj5OLeV03" index.js:33
"SERVERLOG: "socket connect from ID: 4jIA0hgJal5RfsMtr0z8LjYmj5OLeV03"" index.js:79
"SERVERLOG: {}" index.js:79
"SERVERLOG: "Setting up LED as an output on GPIO 18"" index.js:79
"SERVERLOG: "Exporting gpio18"" index.js:79
"SERVERLOG: "Setting up Button as an input on GPIO17"" index.js:79
"SERVERLOG: "Exporting gpio17"" index.js:79
"SERVERLOG: "Setting direction \"out\" on gpio18"" index.js:79
"SERVERLOG: "Setting direction \"in\" on gpio17"" index.js:79
"SERVERLOG: "Current direction is already in"" index.js:79
"SERVERLOG: "Setting direction \"out\" on gpio18"" index.js:79

jmstriegel, 2015-02-25:
I noticed this issue recently as well. I think its a timing issue where the
gpio permission updates lag behind when the device is exported.

I've uploaded v0.9 to the site, which should try to work around the
problem. Can you try out the new image and let me know if that fixes
things? Sorry for the hassle.

bmjbmj, 2015-02-25:
Hello!

Thank you for your time and effort! The new image 0.9 works as intended.

//BMJ

Den 2015-02-25 04:11, Jason Striegel skrev:

closed 2015-03-05 by jmstriegel