googlecreativelab/coder #95: Wifi isn't showing up to connect via PC

WhovianSupreme, 2015-03-04:
So today I tried to install Coder onto my nice little SD card that I had been running Raspbian on. I had a bit of trouble trying to figure out how to get the disk image onto the SD card, since I am running Windows 8.1, and I don't get the nice little installer.
So here is what I did:
Downloaded Coder (the zip)
Downloaded and installed the Bonjour Print Services thing
Downloaded and installed SD Formatter
Formatted SD with Quick Format (just used the setting that it opened with)
Downloaded and installed Win32DiskImager
Extracted the coder disk image out of the folder in the zip that WASN'T the Mac folder (the self-titled one)
Wrote the disk image to the SD card, it said it worked. It also renamed my SD card Coder.
Ejected
Stuck the SD card in my Pi, stuck in power cord, and HDMI cable/cord thing.
There wasn't anything showing up on the screen, so I took out the HDMI cord
And It looks like its on, the little LEDs are on, but the wifi isn't showing up.
I do have a WiFi usb thing, it worked when I was running Raspbian.
Anybody have any idea of what I did wrong? It isn't like the instructions for Windows were super detailed or anything, so....
Frankly, I am a bit frustrated that I couldn't find support anywhere else but here.
Thanks!

whinge, 2015-03-04:
All windows 8 problems still on 8.1.
   It is Microsoft not liking all things  Raspbian and image formatter as
it is still wanting to make Money.
Not give things like Linux Raspbian a chance to grow as it is competion.

I have split my Hard Disk on my laptop, one side is windows 7 and the other
is Linux ..... Arduino and Raspbian work fine.    So windows 8, and 8.1 is
fit for the trash can.
It is not for helping you to use Raspbian, the size of your SD card will
make no difference either.  4 to 64 GB windows is the problem,  go back to
windows 7.    Or perhaps wait for Windows  10  to have it's own problems
sorted. You will be banking your head with 8 & 8.1.
Good luck

jmstriegel, 2015-03-04:
@WhovianSupreme sorry to hear you're having difficulties. The good news is that Coder runs on top of raspbian, so if you've had rasbian working in the past, we should be able to get Coder working as well. Once you get the device booting, it shouldn't matter what OS you have on your computer.  Here's some quick thoughts:

- you saw that the sd card was renamed to coder. that's a good sign and means win32diskimager is doing its job. can you confirm that your sd is 4GB or larger? if not, it's possible for the boot drive to be successfully created (the disk you see mounted in windows), but for there not to be enough disk for the entire linux partition to be created.

- it sounds like you have lights blinking, which is a good sign that the device is actually booting. you should definitely see something on HDMI out, though, if you're using it. in that respect, it should boot just like a normal raspbian image. make sure hdmi is connected and tv is on before you plug in the power -- i've noticed that sometimes the TV doesn't get detected correctly if it's plugged in afterward. If we can get this part working, it might help with debugging what is happening.

- depending on what device you have (B, B+ or 2 B), it might be a half minute to a minute for it to finish booting. at the end of this, if your wifi adapter has an led, you should be able to see it blinking when it establishes a network connection.

- coder will only start its CoderConfig network if it doesn't have an ethernet connection. when there's no ethernet and it hasn't been configured for your wireless network, it will create an Ad-Hoc (device to device) network.

I noticed this article that makes it sound like windows 8.1 won't show ad-hoc networks anymore, and you may need to do something more manual:
http://answers.microsoft.com/en-us/windows/forum/windows8_1-networking/windows-81-ad-hoc-not-showing/5d75d887-ff59-425e-a480-9f80ada63627

If that's true, I'd actually recommend just setting up your wireless connection manually. If you look on the sd card inside coder_settings/reset_files, you'll find a file called wpa_supplicant.txt. Make a copy of that and put it in coder_settings/wpa_supplicant.txt. Edit that file using a program like Notepadd++ that respect unix line endings, and look for a section like this:

```
#
# A typical encypted network using WPA
#
network={
    ssid="Network Name"
    psk="yourpassword"

    # Change to 0 to enable
    disabled=1
    scan_ssid=1
    priority=4
}
```

You can edit that with your network id, password, and change disabled to 0. Save it. When coder boots, it should read this file and start up on your normal wireless network.

WhovianSupreme, 2015-03-04:
@jmstriegel So I just got back to try to fix it, and as soon as I stuck the SD card in my PC it said there was something wrong, and I should scan it. I opened file explorer and it seems as though it should have worked, it seems intact. I just did the whole copying the wpa_supplicant.txt file, and I noticed that it said that I had 55.9 MB total on the SD card. I am a bit confused considering that it said that I had, like, a gig, earlier. So I am not entirely sure what happened there, but I have a sinking feeling that it was because I didn't do the whole Erase All the Things with win32diskimager, and I didn't do any special partition-y stuff. So should I just re-do the whole process and try with the Erase or Overwrite option?

WhovianSupreme, 2015-03-04:
Seems to be working... But after all this trouble, the biggest problem left is that I can't find my mouse.

WhovianSupreme, 2015-03-05:
@jmstriegel Its asking for my login, which I don't have. I tried what I copied with that .txt file earlier, but it didn't work.

jmstriegel, 2015-03-05:
Ok, so it's booting now! Is it asking for that on your TV or in the web browser? You will want to connect from your browser to get things set up for the first time.

If you haven't created a password yet, the console/ssh password is just "raspberry", but if you connect via your web browser the first thing it will want you to do is set a new password.

WhovianSupreme, 2015-03-05:
TV, and I was booting up the Pi, and it asked for coder login, then password.
So what would the login be?

jmstriegel, 2015-03-05:
Until you set it up, login is pi, username is raspberry.

Can you tell if if connected to your wifi network? Instead of logging in at the TV, to connect to coder, you'll want to open chrome on your laptop and go to http://coder.local (or for the moment you can connect to the ip address that you see on your tv).

jmstriegel, 2015-03-05:
Just to clarify, what Coder does is make it possible to program web apps on your pi through a web browser, without it needing to be connected to a keyboard or a tv.

WhovianSupreme, 2015-03-05:
The Pi itself (connected to my TV and a mouse and keyboard) was asking for my login. The network wasn't showing up.

WhovianSupreme, 2015-03-06:
@jmstriegel So I logged in on my Pi, via the TV, and it worked with pi and raspberry as login and pass. However, the wifi isn't showing up. Is there something I have to do to on top of this? And it *is* a seperate wifi called coderconfig, right?

jmstriegel, 2015-03-18:
@WhovianSupreme see my note above about windows 8.1 not working with ad-hoc networks. You aren't going to see any of the Coder stuff while you're looking at things on your tv. I had you connect to the TV so you can confirm that everything is booting, and it also will be a way for you to manually configure things.

I'd recommend following the instructions above to manually configure the device's wireless settings so that it can connect to your router. Just to get stuff working the first time, you can also plug Coder right into an ethernet port on your router. Once it is running on the network, you should connect to the device's IP using your PC's web browser. Usually, this means going to http://coder.local/ but you can also go to http://[ipaddress], where [ipaddress] is the ip address you see printed out when you boot while connected to your TV.

WhovianSupreme, 2015-03-19:
@jmstriegel Are you referring to the article linked above? If so, I tried this one:
>Because the wireless network picker in Windows 8.1 doesn't show ad-hoc networks, connecting must be done by hand.
>To see all networks in range, including ad-hoc, run this command in command shell (cmd.exe):
>> netsh wlan show networks

>SETUP
>This must be done once per network.
>go to "Network and Sharing Center"
>click "Set up a new connection or network"
>double click "Manually connect to a wireless network"
>enter the SSID of the ad-hoc network (as shown by "netsh wlan show networks") into the "Network >name" field
>configure security settings accordingly
>uncheck "Start this connection automatically" (important)
>click "Next", then "Close"
>Run this command (important):
>> netsh wlan set profileparameter <ssid> connectiontype=ibss
>Replace <ssid> with the SSID of your network.

>CONNECT
>After setting up, run this command whenever you want to connect:
> netsh wlan connect <ssid>
>Replace <ssid> with the SSID of your network.
>Hovering over WiFi icon in system tray will show the name of the network that you are currently >connected to, even if it's an ad-hoc network (the network picker will not show it if it's ad-hoc). The >name is also visible in "Network and Sharing Center" window.

And, well, it didn't work at about the first step. I don't see a place anywhere to set up a new connection, it only has Manage Known Networks, Ethernet, and Add a VPN connection.
That's all I see, other than turning off the wifi. Also, (this is shameful) I don't have a clue where the router is, and if I did, I wouldn't know how to set up Ethernet.
I'm terribly sorry for bothering you and using up your time.