googlecreativelab/coder #96: Is there a way to autorun an app made with Coder after auth?

miniBloq, 2015-06-21:
I would like to run one of my apps (made with Coder) after user's authentication on the auth screen. What should I modify? Thanks!

jmstriegel, 2015-06-23:
Interesting... well, you could actually edit the auth program itself, and have it forward to /app/your_app instead of the /app/coder.  "auth" is a normal coder app (just like almost everything in coder), but it's hidden from view on the home screen.

After logging in, if you go to /app/editor/edit/auth, it should open the auth program for editing. Look in the node tab, somewhere around line 122, there's a method called exports.index_handler. Inside, you'll find a line that looks like:

    res.redirect('/app/coder' + firstuse);

Comment that one out, copy it, and change it with your app id:

    //res.redirect('/app/coder' + firstuse);
    res.redirect('/app/your_app' + firstuse);

You'll want to tread lightly, obviously, because you wouldn't want to break how the auth program works and lock yourself out accidentally. :)

Once you've saved, you should be bounced to that url after logging in, instead of the /app/coder url.

miniBloq, 2015-06-23:
Hi! Thanks.

I tried it (in fact I have edited that file before posting my question, only that I did it with nano using SSH). Something seems to not be working, since I'm still redirected to coder. Now I went to https://10.0.0.7/app/editor/edit/auth, edited it in the Node tab and reseted the RasPi (just in case). Here is my code, maybe I'm still missing something:

exports.index_handler = function( req, res ) {
    
    var firstuse = "?firstuse";
    if ( typeof( req.param('firstuse') ) === 'undefined' ) {
        firstuse = "";
    }
    
    if ( !exports.isConfigured() ) {
        res.redirect('/app/auth/configure?firstuse');
    } else if ( !exports.hasPassword() ) {
        res.redirect('/app/auth/addpassword?firstuse');
    } else if ( !exports.isAuthenticated(req) ) {
        res.redirect('/app/auth/login' + firstuse);
    } else {
        //res.redirect('/app/coder' + firstuse);
        res.redirect('/app/thingscoder_v0_52' + firstuse);
    }
};

jmstriegel, 2015-06-23:
Oh right - this fixes half the problem (if you go to / and you are logged in, it should take you to your app now). The other half is the javascript handler for when the login button is clicked.

Look in the JS tab in the auth app. Around line 244 of the loginClick function, you'll see this:

    window.location.href="/app/coder" + firstuse;

You'll just want to do the same thing with that one.

miniBloq, 2015-06-24:
Thank you! It worked!