googlecreativelab/coder #97: How to share ?

nathan30, 2015-09-23:
Hi,

I want to access to my Coder from anywhere, not just on my house

How can i do that ? Is it possible ?

Thanks !

whinge, 2015-09-23:
What is a coder ?  I use Arduino UNO ,mega,micro, Yun, Leonardo,  you need
wifi to make this work and only the Yun has it ,  you can buy a shield with
wifi but I am in the UK !
So it may be different where you are?
That's all I know

nathan30, 2015-09-23:
When I said Coder I mean the environment ^^

I have RPI, and a wifi dongle

So, how to access from anywhere ?

whinge, 2015-09-23:
Find your wifi source mine is in the cloud as in Sky cloud at home I have A
Router which is from SKY.   When I go in to the town centre I log in to the
cloud at selected places Costa Coffee or A large Pub Restuant chain.   So
you need hot spots to get your dongle working.

First you need to know MAC address and then 199.1.12.01 for example as a
web address.
When you have dongle working at home in communication with home router you
can find out
The rest.
If you know the above information you should be able to log on to hot spots
or cloud ports.
I am not an expert so why you are asking me I have no idea where did you
get my email?

MrJoshFisher, 2016-04-03:
Set the ports on your router to forward to your raspberry pi then use a dynamic end service to point a unique address to your raspberry pi, thus accessing it from anywhere

whinge, 2016-04-08:
What is a dynamic end service?

MrJoshFisher, 2016-04-08:
Sorry meant dynamic dns service like noip or duck dns

Josh Fisher