googlecreativelab/coder #102: Does not work on Raspbian Jessie

whoismezero, 2015-10-28:
After running all of the setup scripts and rebooting I cannot change the initial password.  On the first run of the webpage it gives  a sudo error.
[image]

whinge, 2015-10-28:
> <https://cloud.githubusercontent.com/assets/824333/10805687/a8be3732-7d9d-11e5-859b-ff3a7a8d7113.jpg> Same

I just gave up on Linux .same sudo password problem

jmstriegel, 2015-11-02:
Ugh. Weird. Looks like there's something wrong with sudo in Jessie.  I'll need to look into this when I build a new image.

whinge, 2015-11-02:
I shall not bother with Linux!  On Yun or on Windows Laptop!