googlecreativelab/coder #107: Chromebook " issue.

paulmnguyen, 2016-04-04:
When connected on a Chromebook. The kids can't use double quote in the editor. Chromebook does &ldquo; instead of ". Work around for now is they can only use single quotes. Not sure if there is a feature in Chromebook chrome browser that can be changed to fix this.

jmstriegel, 2016-04-04:
Strange -- I haven't seen that before. What keyboard layout are you using? I'm curious if one of the international layouts maps the double quote key differently. Perhaps there is a different keyboard combination you need to use to get a straight double-quote?

paulmnguyen, 2016-04-04:
The classroom is using samsung chromebooks. I don’t believe the kids changed the keyboard layouts. Although it is possibility. I will have to check and let you know. We are all on spring break now.

Thanks,

Paul