googlecreativelab/coder #110: Raspberry Pi 3

miniBloq, 2016-05-20:
Hi, is is possible to run Coder with the RasPi3 and its embedded WiFi?
Thanks!

Enkerli, 2016-06-09:
Sounds like there are instructions here:
https://github.com/googlecreativelab/coder/wiki
(Haven’t tried Coder in any form, yet. Just noticed this issue while checking how active the project was.)

w1jp, 2016-06-15:
Did it work?

miniBloq, 2016-06-15:
I wasn't able to test yet.

w1jp, 2016-06-15:
Okay, I have tired the instructions above. Note that the scripts have changed such that the instructions were different after the first command. IOW-- I didn't do the last two commands as the scripts appeared to do that for me.

However, after following the instructions I was left with nothing to do (i.e., no further instructions). I just assumed all I had to do was reboot. 

After rebooting, the system booted as normal. When I logged in as 'pi' I noticed that the Wi-Fi as down (as you would expect if it is hosting CODER). I looked on my Wi-Fi and I did not see a "CoderConfig" WiFi to connect to.

Now what?

RRoeLAN, 2016-07-12:
I also tried to run it on the model 3, after the command:
`chmod +x coder_bootstrap_install.sh` 
The commands did not work. But the Pi itsself also suggested commands. These commands resulted in a non-booting RPI.

The-Alchemist, 2016-09-21:
Can confirm problems: the 0.9 image doesn't seem to boot on a Raspberry Pi 3.

No "CoderConfig" network, and even the ACT LED doesn't seem to blink.

manuti, 2016-12-07:
The better way I find to run Coder on new Raspberry Pi 3 under Raspbian is install **docker** and run inside a *container*:

	sudo  apt-get install docker.io

And after this, run the image prepared by resin.io

	sudo docker run -d -p 8081:8081 resin/rpi-google-coder

Now you can connect by web browser to `https://your-rpi-3-IP:8081`
Of course you need to learn a little about docker to start, stop and restar the container.

Mycorrhizae, 2017-01-10:
I tried the 0.9 image, then the wiki commands (could not complete because changing of ownership was not permitted,) and finally i tried to download the docker as suggested in the most recently preceding comment and was unable to download because it kept timing out. Can we expect a coder update for RasPi 3 any time in the near future?

manuti, 2017-01-12:
I created a post in Spanish about using Docker and you can see Coder running on top of an armbian intall on Orange Pi One board. https://raspberryparatorpes.net/proyectos/docker-para-torpes-con-raspberry-pi-u-orange-pi/
I know is in Spanish but is more or less easy to follow the steps.

eskwire, 2017-01-12:
@manuti don't call him torpe. You are mean. :p

manuti, 2017-01-12:
sorry @Mycorrhizae and @eskwire I don't want to call torpe to anybody ... is my blog's name! Because when I started 4 years ago I'm a real dummy and "torpe". ^__^

eskwire, 2017-01-12:
@manuti todo bien am the torpe.

neilbalch, 2017-05-06:
This has been broken for some time now. _Could someone who know how to fix it please fix it?_

The-Alchemist, 2017-05-06:
@neilbalch : You can just install the project manually (although that's kind of a pain too).

manuti, 2017-05-08:
@neilbalch as I say before here https://github.com/googlecreativelab/coder/issues/110#issuecomment-265422214 
This is another orphan project by G**gle