googlecreativelab/coder #114: Installation Error Using Mac Installer

PapaLazarou, 2017-01-16:
Hi,

I'm trying to install Coder to an SD card using the CoderSetup installer supplied with the Mac download.  It gets as far as asking for my Mac password, then immediately after I get the error message 'There was an error while writing to your SD Card'.

I've tried multiple SD cards - all to the required specification but with no luck.

I'm not confident enough yet to try the manual installation guides so any help would be appreciated.

Thanks

Craig

CVBruce, 2017-01-16:
I don’t know what is happening, but I’ve stubbed my toe on various occasions and perhaps one of these issues is the problem.
1. reboot your mac
2. Are you able to mount your SD card and copy a file to it.
	a. bad sd to usb adapter <— try a different one
	b. bad/mismatch micro sd to sd adapter <— This one gets me all the time.
	c. write protect tab on SD card in locked position.
	d. bad sd slot on mac. <— Possible, but I haven’t had one, try using a usb adapter.
3. Using “/Applications/Utilities/Disk Utility” can you format, erase, or partition your SD card.
4. Bad password / Account is not an Administrator.  <— Not likely, but included for completeness.

I hope this list helps

PapaLazarou, 2017-01-16:
Thanks Bruce - I've checked all of the above before posting but still have the same issue.  Going to be brave and attempt the manual installation process!

Thanks

Craig

PapaLazarou, 2017-01-16:
Quick update...just done some further research and have found a couple of threads saying that Coder doesn't work on the Pi 3 (apparently the image available to download hasn't been updated to work with the 3).

Does anyone know if this is correct pls?

Thanks

Craig

CVBruce, 2017-01-17:
Something else I thought of.  If you have a Raspberry Pi 2 or 1, download and install the code on that board, then apt-get update && apt-get upgrade && rpi-update, then move the sd card to the Raspberry Pi 3, and see if it works.