googlecreativelab/coder #116: Raspberry pi 3 not booting up with coder

keytocode, 2017-04-13:
Basically, only the red light shows up when I put coder on the sd card then into the pie. No green light indicating it booted up.

neilbalch, 2017-05-06:
You are currently supposed to download raspbian and then download the source and run it manually, [see the bottom of this page](https://github.com/googlecreativelab/coder/wiki). It currently doesn't work, and I am not sure how to make it work.

kamal007007007, 2017-08-23:
coder did not support Raspberry Pi 3
so it is not possible to use google coder on RPi 3

DRD161, 2017-09-16:
Please make Coder compatible with Raspberry Pi 3 or better yet on PC.

neilbalch, 2017-10-10:
This has been a forgotten Google Project for a while now, @DRD161 ..... No one maintains it...

neilbalch, 2017-10-10:
Any hope of this being bodged again vanished with @jmstriegel

Luuii, 2018-10-02:
Who joins to find a way to get the right compatibility the Coder on the Raspberry pi 3. @neilbalch @DRD161 @kamal007007007 @HalfdanJ @cmatta @jmstriegel

BenDoesStuff, 2021-12-02:
Still no fix?