googlecreativelab/coder #117: installed bonjour print services still not able to access http:coder.local

kamal007007007, 2017-08-23:
installed bonjour print services still not able to access http:coder.local and i use to connect my pc to internet with Wi-Fi adapter and raspberry pi 2 to my PC through Ethernet and check in  ''allow other network users to connect through this computer's internet connection'' sharing option are PC & PI will consider to be on same network or is there other issue please help

Vadeghan, 2017-08-23:
What you may be able to do instead is put in the Raspberry Pi's IP address and the port number, separated by a colon, into the address bar. For example - https://10.1.1.10:8080. The default port number should be 8080.

kamal007007007, 2018-01-25:
It worked thanks